ursoft/Marlin #3: При вставке флешки предлагать распечатать самый новый файл

ursoft, 2019-11-19:
В 99% случаев это именно то, что надо. Если пользователь нажал автоматически выбранную кнопку "Печатать", начинаем печать. Если "отмена" или файлов нет, ничего не делаем. Если вышел небольшой таймаут, показываем список файлов.

ilmen, 2019-11-20:
Если новых файлов несколько - список файлов

ursoft, 2019-11-20:
Довольно сложно выбирать, какие файлы новые, а какие - нет, чтобы посчитать, что их несколько и понять, что последний не обязательно хорош. Кто-то слайсит как метеор, у него разница будет в секундах. А кто-то по несколько минут подбирает параметры. В этом сценарии (если порядок важен и требуется выбор), лучше не соглашаться с предложенным последним по дате файлом, а подождать короткий таймаут.

ursoft, 2019-11-21:
Посмотрел исходники - куда проще просто вывести список файлов на вставку карты, и автоматом подсветить последний в Fat (сортировать по дате не будем - это памяти требует; кому порядок не понравится - пусть отформатируют свою карту, и будет всё ОК). При этом отключенное подтверждение не включаем. В итоге получаем одно нажатие, как и изначально. Из минусов - не видно полное имя файла. Если правильно подойти к наименованию, то это не сильный минус. Скроллинг делать не хочется, 2004 с ним плохо справляется.

commit 2bff7397 (2019-11-21): fix #3
closed 2019-11-21 by ursoft