ursoft/Marlin #5: Печать с внутренней карты MicroSD

ursoft, 2019-11-19:
При USB-соединении с принтером доступна эта карта. Скорость записи почти мегабайт в секунду. Возможно, захочется gcode туда засылать. Осталось заставить оттуда распечатать - SDCARD_CONNECTION ONBOARD, но так, чтобы LCD не потерять.

fixid-pub, 2019-11-20:
Это плохая идея, внутренняя карта нужна для обновления прошивки. Туда достаточно записать firmware.bin для автоматического обновления. Octoprint может обновлять прошивку путем монтирования карты и закидывания бинарника прошивки

ursoft, 2019-11-20:
если сломается обновление прошивки, то ничего делать не будем, конечно

ursoft, 2019-11-21:
Обновление прошивки не ломается, но и одновременной работы не предусмотрено. Т.е. если таки доработать marlin для работы с 2 картами, то подав M21/M22 можно внутреннюю смонтировать/отмонтировать. Когда отмонтирована - доступна через USB с ПК. В любом состоянии можно запускать печать из pronterface (сам смонтирует). Самый большой минус - в прошивке это предусмотрена работа с 2 картами, а если и предусмотреть, это скушает немало памяти и работы не так мало, как в других запросах. Так что, возможно, и не стоит овчинка выделки. Время покажет.

ssavickiy, 2019-11-22:
имхо, я не вижу смысла в этой фишке.

ursoft, 2019-11-22:
Такая функция у меня в стоковом дримере была. Прямо из слайсера файл отправлялся по USB на ту флешку, что в материнке торчит (и тут же сам запускался). Удобно, комп можно выключить и никаких тебе расходов на малинку или дерганий SDCARD из разъёма в разъём. Тут же я наверное таки проленюсь это самолично делать (к тому же мне-то малинка приедет скоро, а остальные приучены к своим способам, а не избалованы, как я). Я надеюсь, в исходном репозитории марлина рано или поздно это сделают, раз техническая возможность имеется - может тогда и подтянем.

ssavickiy, 2019-11-22:
@ursoft малинка умеет на сд карту кидать, но как дальше печать происходит не проверял. наверное можно тогда и малину отрубить

ursoft, 2019-11-22:
@ssavickiy если она кидает файл командами GCODE через эмуляцию USB<->RS-232 - то да, можно ее потом отключить. Но наверное лучше все же с нее и печатать, чтобы избежать лишней задержки на предварительную передачу файла по USB-каналу 100-200кБит. У дримера-то был полноценный по скорости канал на внутреннюю флешку - а тут будет куцый на внешнюю.

ssavickiy, 2019-12-08:
@ursoft столкнулся с тем, что все таки это опция должна быть включена SDCARD_CONNECTION ONBOARD
для того, чтобы была возможность через октопринт прошивать плату. но я не пойму, при этом поломается печать с сд карты экрана?

ssavickiy, 2019-12-08:
update
сейчас прошился через октопринт и без этой опции. хотя в самой инструкции указано https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#lpc1768-boards

ursoft, 2019-12-09:
В инструкции опечатка, они должны были написать LCD вместо ONBOARD (это действительно значение по умолчанию). Сами же потом монтируют диск по USB. Если таки доделаем запрос, то октопринт сможет на высокой скорости записывать задание на печать по USB, а потом хоть помирать.

ursoft, 2019-12-09:
Вот какой получается алгоритм (включается SDCARD_CONNECTION=LCD_AND_ONBOARD и предполагает наличие внутренней карты, иначе не имеет смысла):
1. Монтаж карты осуществляется в следующем приоритете: внешняя карта важнее, но если ее нет, смонтируется внутренняя.
2. На старте монтируем карту согласно п1, чтобы проверить, есть ли auto#.g или recovery. Если есть, отрабатываем их и по возможности демонтируем внутреннюю карту (чтобы была видна в ПК как USB-диск). Иначе обязательно демонтируем внутреннюю. Если не демонтировалась, это можно сделать командой M22 (или из меню) - и можно заливать с ПК прошивку или gcode.
3. Если какие-то действия пользователя привели к монтажу внутренней карты, автоматически она потом не демонтируется.
4. Если какие-то действия пользователя привели к монтажу внешней карты, автоматически демонтируется внутренняя.

ssavickiy, 2020-01-23:
Firmware Updater перестал работать на октопринте, причин пока не вижу

ursoft, 2020-01-23:
Забыл написать про PrivateMounts

ursoft, 2020-01-25:
Откатил временно, костыль конфликтует с включением режима UART

closed 2020-01-25 by ursoft

ursoft, 2020-01-28:
теперь порядок

commit 5749521b (2019-12-09): fix #5
commit 3d4507f0 (2020-01-25): fix #5