ursoft/Marlin #6: Длинное нажатие в главном меню

ursoft, 2019-11-19:
При наличии SD карты и скрипта LONG.gcode на ней, выполнить его. Иначе предварительный нагрев PLA.

ilmen, 2019-11-20:
Почему именно PLA?
Может переход в меню выбора профиля преднагрева?

ursoft, 2019-11-20:
Наверное конкретное действие будет выбираться в конфигурации, в моей будет самое частое для меня. Если мне открывать меню, то для завершения операции требуется еще куча действий - теряются несколько секунд и надо смотреть на экран. А так я нащупаю кнопку и зажму, она пискнет.

ursoft, 2019-11-22:
Итого вот что получается: долгое нажатие от двойного отличать там не умеют, а мы обучать не будем. При печати это занято на BabyStepping - и так останется. А вот при ожидании начинается наша петрушка:
1. Ищет в корне флешки файл long_c.g и, если найден, выполняет его
2. Если файла нет, издаёт мерзкий писк и ждет после него секунду. Если Вы попали сюда случайно, до истечения этой секунды Вам следует испугаться и отжать кнопку. В этом случае ничего больше не произойдет.
3. Кнопку держите - значит, знаете, зачем. Со вторым писком можете ее отжать уже. Если после второго писка идет третий - значит, Вы включили предварительный прогрев под первый материал. Если третьего нет - значит, у Вас до этого был некий прогрев - а теперь Вы его выключили.

commit 90786e15 (2019-11-22): fix #6
closed 2019-11-22 by ursoft