ursoft/Marlin #11: EEPROM и PID

ursoft, 2019-11-21:
Изначально была доступна программа минимум: откалибровать через pronterface PID экструдера и записать его в стартовый код слайсера. Но вообще-то марлин умеет это делать из меню и сохранять в EEPROM. Надо попробовать включить этот функционал как для экструдера, так и для стола - и посмотреть, что получится.

ssavickiy, 2019-11-21:
авто пид в меню, а если можно, то с выбором необходиой температуры сделать - пиды у всех разные.

ursoft, 2019-11-21:
сделал, тестирую

ursoft, 2019-11-21:
При включении PIDTEMPBED становится возможно подавать команду `M303 E-1 S80`, но при большом количестве повторов она у меня стала поднимать температуру выше 100°, а при малом предлагает все коэффициенты занулить:
```
SENDING:M303 E-1 S80 C2
PID Autotune start
 bias: 45 d: 45 min: 79.92 max: 83.25
 bias: 41 d: 41 min: 79.89 max: 80.68PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define DEFAULT_bedKp 0.00
#define DEFAULT_bedKi 0.00
#define DEFAULT_bedKd 0.00
```

Поскольку претензий к нагреву стола у меня нет, по столу оставляем все как было.

ursoft, 2019-11-21:
Вроде протестировал, вопрос у меня остался только один: почему для этой команды не предусмотрено никакого GUI - ни прервать процедуру (не перезагружая принтер), ни посмотреть результат (не подключая консоль) - если что-то пошло не так.

ursoft, 2019-11-23:
Надо было пониже температуру: M303 E-1 S70
```
SENDING:M303 E-1 S70 C5
PID Autotune start
 bias: 142 d: 112 min: 67.97 max: 70.24
 bias: 152 d: 102 min: 68.58 max: 70.44 Ku: 139.20 Tu: 10.01
 No overshoot
 Kp: 27.84 Ki: 5.56 Kd: 92.89
 bias: 152 d: 102 min: 69.09 max: 71.01 Ku: 135.76 Tu: 10.03
 No overshoot
 Kp: 27.15 Ki: 5.41 Kd: 90.78
 bias: 130 d: 124 min: 69.74 max: 71.73 Ku: 158.48 Tu: 12.77
 No overshoot
 Kp: 31.70 Ki: 4.96 Kd: 134.94
PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define DEFAULT_bedKp 31.70
#define DEFAULT_bedKi 4.96
#define DEFAULT_bedKd 134.94

```

ursoft, 2019-11-24:
Запускаю модель со столом 60, он догревает до 50 и потом вылетает на ошибку. В прошивке результат автокалибровки на т-ре 80
  #define DEFAULT_bedKp 32.95
  #define DEFAULT_bedKi 5.77
  #define DEFAULT_bedKd 125.32
Если автокалиброваться на 60 градусов, то коэффициенты другие:
#define DEFAULT_bedKp 31.29
#define DEFAULT_bedKi 6.03
#define DEFAULT_bedKd 108.27
и с ними он стол доводит до 60. Короче, ну его в пень, пид стола.

ursoft, 2019-11-25:
Новая информация - оказывается, марлин умеет при включении PIDTEMPBED считывать нули из EEPROM. Достаточно сбросить их, и шарманка начинает греть, как надо.

commit 9e56316d (2019-11-21): fix #11
commit 53988749 (2019-11-23): fix #11
commit 61db3069 (2019-11-24): fix #11
commit 24f81cd8 (2019-11-25): fix #11
closed 2019-11-25 by ursoft