ursoft/Marlin #19: включить uart на 2208

ssavickiy, 2019-11-21:
вот пример
https://www.youtube.com/watch?v=sSO3Xd0a0Z8

но были какие-то проблемы с ними на экструдере и/или с включенным linearadvance

ursoft, 2019-11-21:
тут целое исследование надо провести: какие преимущества получим - станет ли тише, чем в стоке, снизится ли вероятность пропуска шагов и т.д.

kapacb, 2019-11-21:
> но были какие-то проблемы с ними на экструдере и/или с включенным linearadvance

https://3dtoday.ru/blogs/nayuch/linear-advance-skr-13-tmc2208-21301/

kapacb, 2019-11-21:
> какие преимущества получим 

- управление током через меню принтера / по USB ( команда m906 ), а не потенциометром
- расширенные настройки чипа (сначала нужно раскурить datasheet, потом - библиотеку TMCStepper )
---

>  станет ли тише, чем в стоке,

нет (если только не подобрать параметры == see datasheet)
---

> снизится ли вероятность пропуска шагов и т.д.

нет, т.к. это целиком зависит от механики и настройки токов (ну и перегрева чипа)

ssavickiy, 2019-11-21:
сейчас прошивка вообще не знает какие у нее драйвера. если не хотите включать uart на 2208 - то можно просто указать TMC2208_standalone
правда какие это преимущества даст я не знаю) но на 1 марлине я указыва так.

ursoft, 2019-11-21:
Поскольку в кит/принтер идут платы с ненастроенным UART, поставить его в прошивку я могу только для своей личной сборки, да и то после того, как залезу к себе в подвал, это будет (надеюсь) не раньше, чем приедет Raspberry (в декабре). А вот поставить TMC2208_standalone и убедиться, что хотя бы хуже не стало, пожалуй, пора.

ursoft, 2019-11-22:
Стало хуже - на небольшой высоте драйвер перестает управлять двигателем экструдера, ситуация известная. Нужен человек, который возьмет у меня или сам соберет текущие исходники (см. #10 - вариант ursoft), поставит ненулевой LA в gcode (или в меню настроит), установит перемычками режим UART на своем изделии и протестирует на высокой (хотя бы 10мм) модели. После чего напишет подробную статью здесь в вики (для нубов). Я дам доступ и подправлю текст, если что. Чуть позже проверю на своём изделии.

ssavickiy, 2019-11-22:
@ursoft в режиме uart или с tmc2208_standalone стало хуже?

ssavickiy, 2019-11-22:
для экструдера можно просто не указывать режим uart

ursoft, 2019-11-22:
Насколько я понял, нужно как раз указать всем дровам UART, а экструдеру через UART отключить STEALTHCHOP_E. Но без перестановки перемычек это сделать нельзя.

ursoft, 2019-11-22:
> @ursoft в режиме uart или с tmc2208_standalone стало хуже?
@ssavickiy я пока не лез к себе в подвал (и полезу не скоро), поэтому у меня стало хуже в tmc2208_standalone - где не работает отключение STEALTHCHOP_E, так нужное для качественной работы в режиме LA

ursoft, 2019-11-22:
> для экструдера можно просто не указывать режим uart

в этом случае вне зависимости от указанного типа (tmc2208_standalone, пусто) драйвер будет работать в своем режиме по умолчанию - видимо, это STEALTHCHOP. И останавливаться, когда ему надоест дергать пруток туда-сюда.

ursoft, 2019-11-22:
Самое плохое (если я прав) в этой ситуации, что не только LA может привести к отказу драйвера, но и слишком частые ретракты из слайсера.

ursoft, 2019-11-22:
Если что, моё мнение основано не столько на [вышеупомянутой статье](https://3dtoday.ru/blogs/nayuch/linear-advance-skr-13-tmc2208-21301/), сколько на https://github.com/MarlinFirmware/Marlin/issues/14634#issuecomment-536853971 , https://github.com/MarlinFirmware/Marlin/issues/11024#issuecomment-397821195 и https://github.com/MarlinFirmware/Marlin/issues/11024#issuecomment-402426130

ursoft, 2020-05-16:
Драйверы есть, принтеры выпускаются с UART, примеры рабочей конфигурации имеются.
Почему бы уже и не обновиться.

commit 7f3c53ac (2020-06-22): fix #19
closed 2020-06-22 by ursoft