ursoft/Marlin #22: скорость загрузки филамента

ssavickiy, 2019-12-03:
скорость загрузки филамента из меню экрана, думаю можно поднять в два-три раза.

ursoft, 2019-12-03:
В рамках #20 я пытался немного поэкспериментировать. Двигатель шумит, фидер грозит либо развалиться, либо прогрызть дыру в прутке, а сам пруток - сломаться внутри при подаче или застрять морковкой при неаккуратном извлечении. Пока практикую отведение шестерни и ручные движения, но уже надоедает. И тоскую по директу.

ursoft, 2019-12-06:
Сейчас стоит 60мм/с при ускорении 25. По умолчанию поставлена длина боудена=0, если через меню выставить правильную (у меня примерно 800), начинает довольно шустро выгружать и загружать. Главное, чтобы не уперлось и не застряло (может и пруток слизать). Кроме того, при загрузке у меня как доходит до сопла, постукивает пару секунд - видимо, длину загрузки надо поменьше сделать.

closed 2019-12-11 by ursoft