ursoft/Marlin #33: Не сохраняется цвет 12864 (по умолчанию).

ursoft, 2020-03-02:
в Марлине такое сохранение не предусмотрено. Однако существует команда M150, которая устанавливает это значение - т.е. не так уж и сложно было бы сохранять его в EEPROM.

ssavickiy, 2020-03-02:
делал сохранение настроек в епром, после смены цвета, но кажется не сработало. надо проверить

ursoft, 2020-03-02:
не пишутся они, это 100%

ursoft, 2020-03-02:
Добавление их в EEPROM буду делать по символу **#define NEOPIXEL_EEPROM_STORE_HACK** в свободное место (которое оттяпаю у тринамиков - там есть пространство, у экструдеров E4 & E5).

ursoft, 2020-03-11:
onHeatingDone -> leds.set_white
onPrintCompleted -> leds.set_green
onHotendHeating & onBedHeating & onPidTuningDone -> от розового до фиолетового

ursoft, 2020-04-07:
Стабильность подсветки экрана как-то захотелось. Пусть Марлин развлекается с оставшимися светодиодами.

commit d26bc970 (2020-04-09): fix #33
commit 9384f737 (2020-04-09): fix #33
commit 6e351fc5 (2020-05-09): Ssavickiy (#47)

* Add TMC micro-steps sanity check (#17044)

* Suppress thisItemNr warning (#17049)

* Don't define 'valptr' if unused (#17048)

* Fix ambiguous type

* fix #33

* Fix G34 probing range/error bug (#17052)

* Default on/off for Power Loss Recovery (#17051)

* Fix nested comment warning (#17045)

* Keep name around for old configs

* Finish M900 updates

* Add single extruder LIN_ADVANCE test

* Neopixel on LPC uses GPIO toggling

* [cron] Bump distribution date (2020-03-03)

* Fix MMU test

* fix #34

* No more direct G28 calls

* Asynchronous M114 and (R)ealtime position optio [... cut]
commit 397bbce4 (2020-05-23): Ssavickiy (#49)

* Suppress thisItemNr warning (#17049)

* Don't define 'valptr' if unused (#17048)

* Fix ambiguous type

* fix #33

* Fix G34 probing range/error bug (#17052)

* Default on/off for Power Loss Recovery (#17051)

* Fix nested comment warning (#17045)

* Keep name around for old configs

* Finish M900 updates

* Add single extruder LIN_ADVANCE test

* Neopixel on LPC uses GPIO toggling

* [cron] Bump distribution date (2020-03-03)

* Fix MMU test

* fix #34

* No more direct G28 calls

* Asynchronous M114 and (R)ealtime position option (#17032)

* Minor EEPROM cleanup

* Softwar [... cut]
closed 2020-04-09 by ursoft