ursoft/Marlin #44: Реализовать прогрессбар на неопикселе

DrOnJD, 2020-03-30:
Возможные фичи к реализации:

- Цветовые статусы подготовки к печати и окончании печати
- Процентный статус окончания печати по средством заполнения ленты неопикселя.
- Возможность выбора цветов для тех или иных статусов

ursoft, 2020-03-30:
Первое и третье в некотором смысле Марлин уже имеет (только [цвета закодированы](https://github.com/ursoft/Marlin/issues/33#issuecomment-597796996), т.е. конфигурирование производится перед компиляцией, а не через меню). Даже поддержка адресного неопикселя есть (в дисплее Fysetc Mini 12864 2.1 три светодиода RGBW). Осталось понять, как проще всего соединить ленту с процентом (который, к слову, в марлине тоже есть уже).

DrOnJD, 2020-03-30:
Ну цвета было бы здорово выбирать HEX'ом. Но это просо хотелка, хотя и не беспочвенная.

> Осталось понять, как проще всего соединить ленту с процентом (который, к слову, в марлине тоже есть уже).
Вот это для меня как раз загадка

ursoft, 2020-03-30:
> Ну цвета было бы здорово выбирать HEX'ом. Но это просо хотелка, хотя и не беспочвенная.
найдите в Marlin\src\feature\leds\leds.h комментарий
**Color helpers and presets**
в нем все RGB-запчасти именованных цветов

ursoft, 2020-03-30:
> Вот это для меня как раз загадка

Отгадку вижу в файле `Marlin\src\feature\leds\neopixel.cpp`. Следует использовать метод `set_pixel_color(const uint16_t n, const uint32_t c)` класса `Marlin_NeoPixel`. Общее количество пикселей NEOPIXEL_PIXELS, процент выполнения файла хранится в глобальной переменной `static uint8_t lastProgress` модуля `Marlin\src\lcd\dogm\status_screen_DOGM.cpp` (есть и альтернативные способы его получить - зависит от конфига). 

У меня ленты нет пока что, но написанного достаточно, чтобы Вы уже смогли попробовать

closed 2020-06-22 by ursoft